adafruit/Adafruit_Seesaw #2: Joy Featherwing only returns joystick values on HUZZAH32 with this library

SoftwareGuy, 2018-03-16:
**Configuration:** HUZZAH32 ESP32 Board + Joy Featherwing "jerryrigged" to the back of a LCD featherwing.

**What I expected:**
Joystick X and Y data plus buttons saying they've been pressed.

**What I got:**
Only joystick reports data back, buttons say nothing.

More info:
- By jerryrigged, I mean I have male to female connectors connecting the required pins to the appropriate pins on the joywing. 
- ACT light does light up when polling for data. However, only the joystick works and it turns stuff like 513,516 etc. Pressing any button, be it A/B/X/Y/Select does nothing. 
- Optional interrupt pins are NOT in use.

Possible hardware defect or ESP32 cannot do bitmasks just yet on SDA/SCK?

ladyada, 2018-03-16:
the example code uses the IRQ, please use the featherwing as expected :)

SoftwareGuy, 2018-03-17:
I am pretty sure I am using the example code 1:1 with a minor edit to the naming of the pins. I'll give this another shot.

SoftwareGuy, 2018-03-17:
Nope, double checked and uploaded the code to my ESP32 board. Only joystick input data gets returned on the Serial output, no button presses. Definitely something amiss here.

On the ESP32, pin 14 is also shared with the SDCard CS pin on the feather wing. Is that causing a conflict?

ladyada, 2018-03-17:
ok we can take a look but we *did* test the ESP32 directly with this wing by plugging it in and it worked

SoftwareGuy, 2018-03-17:
Okay, so you say that the board did work directly on top of the HUZZAH32 without anything else attached? I see on your page that you have a demo video of someone pressing the buttons and a small color TFT breakout board showed the inputs. It does look like it was "stacked".

SoftwareGuy, 2018-03-18:
Okay, results are in. I sat down and unplugged the TFT display featherwing, plugged the HUZZAH32 into a bread board and attached wires from the esp32 to the huzzah. 

Joystick info was returned. I checked the buttons. Got "Button A/B/X/Y" pressed.

So, it WAS a conflict with the SD card slot on the TFT Display Featherwing. I request that this finding needs to be documented on the TFT display featherwing or the Joywing page(s) as it *may* potentially catch out other developers. I suppose that the TFT Breakout shield won't be affected.

I am somewhat satisfied that this has been resolved, but I am bummed that it is incompatible with the TFT featherwing.

ladyada, 2018-03-18:
hmm there shouldnt be, the joy wing has all gpio handled in i2c  (other than the irq), so it could be some other issue you've got? you can post in the forums for tech support - but this doesn't seem to be a seesaw bug :)

closed 2018-03-18 by ladyada

SoftwareGuy, 2018-03-18:
Sure, I'll follow it up there.