adafruit/Adafruit_Seesaw #7: TFTShield18 shieldtest example won't compile

caternuson, 2018-08-21:
```
Arduino: 1.8.5 (Linux), Board: "Arduino/Genuino Uno"

/home/user/Arduino/libraries/Adafruit_seesaw_Library/examples/TFTShield18/shieldtest/shieldtest.ino: In function 'void bmpDraw(char*, uint8_t, uint8_t)':
shieldtest:235: error: 'class Adafruit_ST7735' has no member named 'Color565'
             tft.pushColor(tft.Color565(r,g,b));
                               ^
```
This line:
https://github.com/adafruit/Adafruit_Seesaw/blob/master/examples/TFTShield18/shieldtest/shieldtest.ino#L235

**Adafruit_ST7735 -> Adafruit_ST77xx -> Adafruit_SPITFT**

Just a case issue?
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.h#L87

caternuson, 2018-08-24:
Also another small typo in this one:
**Adafruit_Seesaw/examples/RGBTrellis/basic/basic.ino**
```
basic:63: error: 'rellis' was not declared in this scope
   rellis.read();
   ^
exit status 1
```

ladyada, 2018-08-24:
yeah im workin on these :D

caternuson, 2018-08-24:
awesome. yah. just noticed the commits. no need to mention the esp32 fails now :) you're on it, i'll leave things alone, rock on.

ladyada, 2018-08-24:
kk closin' maybe if you're up for it you can test once i get both libraries up and released :)

closed 2018-08-24 by ladyada