adafruit/Adafruit_Seesaw #11: For a clearer documentation - Discussion

Ve2mrx, 2018-08-29:
Hi everyone!
_This discussion is about precision, clarity and usefulness of code documentation._

_This "issue" is not meant to criticize the work done, but to have everyone think about the usefulness of the comments and documentation they write for their code._ 

I'm sure most have seen well written documentation that is clear, precise and useful as well as useless superficial documentation that tells nothing but what everybody already knew or could figure out in five minutes. I, like many others I am sure, appreciate the effort of preparing the code for documentation, but sometimes, the info given is somewhat "not useful"...

An unfortunate example from [here](8a22b331661d53a0c8faf8f2bf2153661f729db7):
```
/**************************************************************************/
/*! 
    @brief  set the value of the backlight
    @param  value the backlight value to set
*/
/**************************************************************************/
void Adafruit_TFTShield18::setBacklight(uint16_t value){
    uint8_t cmd[] = {0x00, (uint8_t)(value >> 8), (uint8_t)value};
    this->write(SEESAW_TIMER_BASE, SEESAW_TIMER_PWM, cmd, 3);
}
```
I mean, I can guess from the function name that it sets backlight and that I should pass it a uint16_t, but what value? It kinda states the obvious, without giving any _new_, _useful_ information.

Searching the source code was of no help, and I could not find the code for the PWM module for Seesaw.  If I had, I could have figured what it wrote to the CPU registers and find what values the chip expected. I searched everywhere I knew and read all I could find on Seesaw, including [here](https://adafruit.github.io/Adafruit_Seesaw/html/index.html).  But I had to raise an issue to get an answer (Thanks a lot BTW!). 

The updated code from the [Adafruit-ST7735-Library\examples\TFTShield18_seesaw\shieldtest\shieldtest.ino](https://github.com/adafruit/Adafruit-ST7735-Library/blob/master/examples/TFTShield18_seesaw/shieldtest/shieldtest.ino) now gives more information than the current documentation:
```
  // Set backlight on fully
  // ss.setBacklight(TFTSHIELD_BACKLIGHT_ON);
  // Or you can set the backlight one third on
  // ss.setBacklight(TFTSHIELD_BACKLIGHT_ON / 3);
  // Or dim it up
  for (int32_t i=TFTSHIELD_BACKLIGHT_OFF; i<TFTSHIELD_BACKLIGHT_ON; i+=100) {
    ss.setBacklight(i);
    delay(1);
  }
```
Yay! I know how to dim the backlight and what values it will accept! THANKS!

But unfortunately, documentation like the following isn't useful for anybody but those who already know what it does. Then why write it? Don't do this please!
```
    @brief  set the value of the backlight
    @param  value the backlight value to set
```
Maybe this documentation is barely in Alpha stage and just the skeleton of what will be the true documentation? In that case, great! I'm looking forward to its completion!

But right now, it's not very useful.

And still, I have no idea what this does nor why/when I should use it:
```
/**************************************************************************/
/*! 
    @brief  set the PWM frequency for the backlight
    @param  freq the frequency to set the backlight to
*/
/**************************************************************************/
void Adafruit_TFTShield18::setBacklightFreq(uint16_t freq){
    uint8_t cmd[] = {0x0, (uint8_t)(freq >> 8), (uint8_t)freq};
	this->write(SEESAW_TIMER_BASE, SEESAW_TIMER_FREQ, cmd, 3);
}
```

As you can see, without good documentation and comments, that code is like a big black box. What is the usefulness of setting the backlight frequency? When should I use it? What are the acceptable values? 

In itself, for someone who can read code, the code is the ultimate documentation. But without the context in the programmer's head, some information can be lost (Why is it done that way?) and fixes go unnoticed (Doing ___ and ___ to avoid ___ bug in ___.). It may take a significant time to find your answer too!

Will you remember why/how/what in five years when you modify your code? Don't take a chance, document your code!

_Finally, I want to thank the whole Adafruit team who made my purchases useful and fun to use! Yes, I had glitches, but they where taken care of quickly and to my entire satisfaction!_ Sure, you could find cheaper stuff out there, but the quality of the hardware and availability of supported code will make your experience enjoyable! Without good quality parts, ready to use _tested_ code and support, how much time will you lose? I won't!

Martin

Ve2mrx, 2018-08-29:
Thank you for everything!

My adventure in C/C++/Arduinos started less than two months ago. Electronics is what I studied in 1992, but life chose another path for my life....
I haven't learned enough about GitHub yet, but pull requests will come when I have more confidence in my newbie programming skills (:-p) and I have something useful to contribute!

Describing Adafruit as a Learning-oriented company with a gift shop is not far from the truth! I learned a lot in all your tutorials, code examples and documentation!

Thanks for helping me enjoy electronics again!
Sincerely,
Martin

ladyada, 2018-08-29:
aww thank you :) we have some great github learn guides - check em out here https://learn.adafruit.com/search?q=github :)