adafruit/Adafruit_Seesaw #15: no brightness setting on feather crickit neopixels

mnelsoneorm, 2018-09-27:
feather M4 on crickit. Circuitpython.

`import time
from adafruit_crickit import crickit
from adafruit_seesaw.neopixel import NeoPixel

num_pixels = 20

pixels - NeoPixel(crickit.seesaw, 20, num_pixels, brightness=0.2)`

Whatever I set the brightness to it runs at full brightness. The playground express on the crickit does work just fine. I'm following the guide "Make It Glow With Crickit". The guide also has brightness set for the circuit playground example but not on the feather example.

deanm1278, 2018-09-27:
ok this is fixed! please try your code again using this release:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/releases/tag/1.2.4

mnelsoneorm, 2018-10-01:
Nice!  Thank you!  Works now.

closed 2018-10-01 by deanm1278