adafruit/Adafruit_Seesaw #18: Potential lock up with many rapid keypresses.

caternuson, 2019-01-10:
https://forums.adafruit.com/viewtopic.php?f=8&t=144551

ladyada, 2019-01-10:
replied to thread, the code is not easy to debug, so not sure if/when i'll get to it. delays may solve the problem

griv, 2020-04-16:
I've just hit this issue.
I've got a single Neotrellis wired to a D1 Mini. My app connects to an MQTT broker using the PubSub library and subscribes to 16 topics. As the subscription messages come in the Neotrellis locks up after 4 messages. I'm not pressing any Trellis buttons, just writing the LED colours. The D1 mini continues as I can still see more messages coming in in the serial console.

Loading the example code of basic and interrupt I can press buttons fin for maybe 10 seconds and then it locks up. I've tried powering it with 5v and 3.3v with similar results.

ladyada, 2021-04-03:
found this bug, https://github.com/adafruit/seesaw/pull/48
i think disabling IRQ support helps/fixes - ironically :(
so comment out `ss.setGPIOInterrupts(button_mask, 1);` or whatever is enabling GPIO irqs
closing for now because im pretty sure its fixed - have been smashing these MX keys

closed 2021-04-03 by ladyada