adafruit/Adafruit_Seesaw #19: Multiple keypress detection issue

toolbits, 2019-02-06:
The other day, I bought some NeoTrellis and started to make my work.
I found a strange operation when I pressed more than one key at once.

The issue is next:
o When I pressed two keys at one, remaining keys on same column are strangely on.
o In basic example code, key events are correctly come (detecting key pressing generates key event. this case 'detecting' include hardware's wrong key press detecting.).
o getKeypadCount() [line 83 of Adafruit_NeoTrellis.cpp] returns 1~4. (I think 1 or 2 is correct. 3 and 4 is incorrect state of this issue.)
o Using INT pin or not has no relation.
o Using KEYPAD_EDGE_HIGH and KEYPAD_EDGE_LOW causes same act.
o There are another issues I found. One is lockup problem, and the other is key release event no coming.

I tested with Adafruit's original example code named 'basic', using Arduino UNO.
See attached image.

[image]

I uploaded a video on my repository.
https://github.com/toolbits/seesaw/blob/master/%7Eresearch/neotrellis.m4v

So, I doubt seesaw library that has some bugs maybe.
I doubt the code nearby next code:

In https://github.com/adafruit/seesaw/blob/master/source/AOKeypad.cpp

226    gpio_outset_bulk(PORTA, keypad_output_masks[i]);
227    //short delay
228    for(int __tmr = 0; __tmr<100; __tmr++) __asm__ volatile ("NOP;");
229    in = gpio_read_bulk() & KEYPAD_INPUT_MASK; //read everything at once

Or interruption problem of ARM chip. Register remembering and restoring problem, etc...

ladyada, 2019-02-06:
hiya the column select is likely this hardware issue - check
https://learn.adafruit.com/adafruit-neotrellis-m4/faq-troubleshooting#faq-6

toolbits, 2019-02-07:
Hi, thank you for your quick reply.

I tested your suggestion and got a good result !
I havn't did software modifying.
I made a hardware fix, molding SMD LED's pin with epoxy glue.

Please see attached photos.

[image]

[image]

So, multiple keypress detection issue was cleared now. That was not a software problem, that was hardware one.

Now, lockup problem and key release event problem are still there.
I will continue to research them.

toolbits, 2019-02-08:
Ok, I found a certain behavior.
When key release event doesn't come, event flow is below:

1. press targeted button with many other buttons
2. SEESAW_KEYPAD_EDGE_RISING of targeted button comes
3. release that button
4. SEESAW_KEYPAD_EDGE_FALLING doesn't come
5. press same button
6. SEESAW_KEYPAD_EDGE_RISING comes
7. release that button
8. SEESAW_KEYPAD_EDGE_FALLING comes

So, It seems that there is a buffer overflow problem of event queue.
I think it will be avoided by shorter the polling interval.

toolbits, 2019-02-08:
Next, I found the reason of lockup issue.
I think when lockup is begining, your finger or other pin is contacting the SDA or SCL land.
It causes some noise on I2C bus line. Then Arduino or other master board goes to freeze.

I masked the back side of NeoTrellis with the masking tape.
Now, lockup issue seems solved.

[image]

ladyada, 2019-02-08:
yes humans are conductive - good work debugging! :)

closed 2019-02-08 by ladyada