adafruit/Adafruit_Seesaw #20: NeoPixel clear() method wont clear pixels

mic159, 2019-02-09:
It only clears the local state, not the buffer on the seesaw.

The code only memsets the local buffer: https://github.com/adafruit/Adafruit_Seesaw/blob/ddd69568db4389ad735df1088d27507a48548d3e/seesaw_neopixel.cpp#L287

And there does not appear to be any command on the seesaw to actually clear it either:
https://github.com/adafruit/seesaw/blob/ad471bd26fd5ddb4b320dafd0275e58e5f052714/include/event.h#L118

What is the best way to clear all pixels?
Looping all pixels and calling `setPixelColor` seems inefficient.
Can we use `NEOPIXEL_SET_BUFFER_REQ` to set all pixels to 0 at once?

ladyada, 2019-02-09:
call show()
https://github.com/adafruit/Adafruit_Seesaw/blob/ddd69568db4389ad735df1088d27507a48548d3e/seesaw_neopixel.cpp#L102

mic159, 2019-02-09:
@ladyada the `show()` function only sends the `SEESAW_NEOPIXEL_SHOW` command to the NeoTrellis/Seesaw, but the buffer on the NeoTrellis/Seesaw still has the original pixel data, so the LEDs stay lit up as they were.

https://github.com/adafruit/Adafruit_Seesaw/blob/ddd69568db4389ad735df1088d27507a48548d3e/seesaw_neopixel.cpp#L114

Simple reproduce case:

```c
#include "Adafruit_NeoTrellis.h"

Adafruit_NeoTrellis pad;

void setup() {
  pad.begin();
}

void loop() {
  pad.pixels.setPixelColor(0, 255, 0, 0);
  pad.pixels.setPixelColor(15, 255, 0, 0);
  pad.pixels.show();
  delay(1000);
  pad.pixels.clear();
  pad.pixels.show();
  delay(1000);
}
```

This should blink the pixels, but it just stays solid.

ladyada, 2019-02-09:
ooooooooooooooh - yeah thats a bug. sorry no fix at this time because the firmware is pre-burned :( you have to write to each pixel!

ladyada, 2019-02-09:
wanna try adding a
`this->write(SEESAW_NEOPIXEL_BASE, SEESAW_NEOPIXEL_BUF, writeBuf, len + 2 );`
type call after the memset in clear()? that should then send that data buffer off, may need adjustment for variable names n such - i dont have a seesaw on hand right now to test

mic159, 2019-02-09:
@ladyada yes, it looks like we can use `SEESAW_NEOPIXEL_BUF` a few times to clear the buffer on the remote end. Unfortunately the buffer is larger than the 32 byte maximum of TwoWire, so it has to be called in a loop.

Check #21

closed 2019-02-11 by ladyada

rEegLer, 2019-10-17:
I don't believe this was ever fixed for Adafruit Neopixel?

commit 134e8560 (2019-02-11): Clear buffer on seesaw as well (fixes #20)
commit 8aea601a (2019-02-11): Merge pull request #21 from mic159/master

NeoPixels: Clear buffer on seesaw as well (fixes #20)