adafruit/Adafruit_Seesaw #24: Interrupt examples don't use interrupts

zostay, 2019-07-12:
It would be nice if there was an example of using `setGPIOInterrupts` with `attachInterrupt`. The existing interrupt examples seem to be slightly misleading in suggesting that you can properly test for interrupts by watching for a `LOW` signal on a local pin that's been connected to the IRQ output on the SeeSaw module. However, what is properly needed is some code that can use `attachInterrupt` to detecting a `FALLING` signal instead.

ladyada, 2019-07-12:
sounds like a good idea, want to try submitting a PR for a new example that has an attachInterrupt line?

zostay, 2019-07-12:
Uh, well... shucks. You see, I could, but I have no way to test it at the moment. I only have Crickit Hats, RPi0s, and CPEs to play with right now. (I'm not going to use this library or Arduino to connect to SeeSaw for this project. I'm reading this code to understand how SeeSaw works so I can port it to Perl 6 for use with the hats.)

ladyada, 2019-07-12:
a CPX can run arduino to test it for sure.

zostay, 2019-07-12:
I suppose. We'll see.

ladyada, 2021-04-04:
closing due to lack of activity

closed 2021-04-04 by ladyada