adafruit/Adafruit_Seesaw #36: touchRead hangs when reading STEMMA soil sensor with ESP32 DEVKIT V1

jimfrankfort, 2021-04-12:
- Arduino board:  ESP32 DEVKIT V1

- Arduino IDE version (found in Arduino -> About Arduino menu):  Arduino 1.6/1.8 in Visual Studio 2019 set for Adafruit ESP feather.  Using Visual Micro extension to Visual Studio.

I wrote this up in the Adafruit forums.  See the last 3 entries by Jim_Frankfort in this thread:  https://forums.adafruit.com/viewtopic.php?f=19&t=162556&start=15, 

In brief: TouchRead doesn not work on ESP32 DEVKIT V1 using Adafruit soilsensor.  Code worked with Arduino Mega 2560 but hung in read() when executed on ESP32.
    Issue seemed to be duration of optional delay between request sent to soilsensor and reading soilsensor.  If delay too brief, device returns 65535 and code hangs because 
    while loop continues to request read until returns something other than 65535, which it never does.
    Solution is to lengthen delay and only try a finite number of times. A delay of 5000 took only a single call to read to get
    a valid result, where as a delay of 2500 took 2 reads, 1000 took 3 reads, and 500 took 4 or 8 reads.  There are other ways that this code will hang.  E.g. if the soil sensor loses power, it returns 65535.  

Instead of using a while loop, I used a for loop with a max of 255 tries before returning.  In addition to tweaking the code so that it doesn't hang, please consider consider adding an error flag.  I'm working on a data logger that has a lot of sensors.....sure would be good for the system to report sensor failures rather than just stop working.  Thanks......jimf

ladyada, 2021-04-12:
thanks for the note, please submit a PR with a long enough delay for the ESP32 and we'll make sure it works on other boards too :)

ladyada, 2021-04-28:
ok try 1.4.1 please!

closed 2021-04-28 by ladyada

naotakak, 2021-05-13:
Arduino board: HUZZAH32
Arduino IDE version: 1.8.13

I was having a similar issue with my board, although the capacitance value that was returned was seemingly random and extremely high (ranging from 30,000-60,000) even though the temperature was correct. I tried with versions 1.4.3, 1.4.2, 1.4.1, 1.4.0 with the same result. When I used Jim's code as posted in the most recent comment on the forum, the capacitance was returning normal values. 
Thank you so much Jim, this problem has been giving me a ton of trouble on my project!

jimfrankfort, 2021-05-20:
Retried ESP32 test of SoilSensor using Seesaw 1.4.3 and it worked....thanks for fixing.  I liked your implementation better!  Potential hang issue resolved when reading, and potential hang for hardware failure on sensor resolved too :-).  

In my test I was getting readings from ~320  to ~ 1010.  When power removed from sensor, got 65535.
Thanks for fixing......jimf