adafruit/Adafruit_Seesaw #37: Adafruit Seesaw Library 1.4.0 possible error with Soil Sensor

thorathome, 2021-04-12:
Hi Adafruit.
**First, thanks. I love the Adafruit libraries** They make me buy from Adafruit.

**You may have an error in Seesaw library 1.4.0 as it applies to the STEMMA soil sensors.**

The recent Adafruit_seesaw library v 1.4.0 causes my 2 STEMMA soil sensors to read radically different moisture values from the values they both showed prior to library version 1.4.0. Temperature values were not impacted.

Seesaw library version 1.4.0 was installed using Arduino IDE 1.8.13 Win 10 Library Manager.
I am compiling for ESP8266 NodeMCU 1.0 (ESP-12E Module).

Using Seesaw Library versions 1.3.0 and 1.3.1, both my STEMMA soil sensors gave moisture readings in the 200-1200 range.
Introducing library v 1.4.0 changed both soil sensor readings to 20-22 from the previous readings around 600.
Reverting back to library versions 1.3.0 or 1.3.1 restored the previous moisture readings at the 600 range they had been for weeks.

Your seesaw library example code works. Here's simplified code I extracted from my own sketch. 

```
#define SERIAL_SPEED      230400
#define SKETCH_NAME       "Seesaw_Soil_Sensor_Example_Simple"

// This from Adafruit doc https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/arduino-test
#include "Adafruit_seesaw.h"           // Also uses Adafruit Unified Sensor library

// Data object for the STEMMA Soil Sensor
Adafruit_seesaw soilSensor;

void setup()
{
  // Never enough good printouts
  Serial.begin ( SERIAL_SPEED );
  delay ( 500 ); 
  Serial.println ( "\n\n=============================================\n=============================================" );
  Serial.println ( SKETCH_NAME );

  initializeSeesawSoilSensor();            // set up all sensors

} //end setup

void loop()
{
  readSeesawSoilSensorReadings();
  delay ( 3000 ); 
} // end loop

void initializeSeesawSoilSensor()  // soil temp and moisture
{
  // This from Adafruit doc https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/arduino-test
  Serial.println ( "Adafruit Seesaw Soil Sensor initialization" );

  if ( soilSensor.begin ( 0x36 ) )   
  {
    Serial.print ( "Soil Sensor found. version: " ); Serial.println ( soilSensor.getVersion(), HEX );
  }
  else
  {
    Serial.println ( "No soil sensor found. " );
  }
  Serial.println(); 
} // end initializeSeesawSoilSensors

void readSeesawSoilSensorReadings()
{
  // This from Adafruit doc https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/arduino-test
  Serial.println ( "readSeesawSoilSensorReadings called..." ); 

  int moisture;
  float temperatureC, temperatureF;

  Serial.print ( "\nReading soil sensor." );

  uint32_t timestamp = millis();
  temperatureC = soilSensor.getTemp();           // Grab the Soil temperature
  moisture     = soilSensor.touchRead(0);        // Capacitive moisture, too (int)
  timestamp = millis() - timestamp;

  temperatureF = temperatureC * 9.0/5.0 + 32.0;        // convert to Farenheit
 
  Serial.print ( "Temperature C: " ); Serial.print ( temperatureC );    Serial.println ( " degrees C" );
  Serial.print ( "Temperature F: " ); Serial.print ( temperatureF );    Serial.println ( " degrees F" );     
  Serial.print ( "Moisture:      " ); Serial.print ( moisture );        Serial.println ( " units" );
 
  Serial.print ( "Read duration (ms): " ); Serial.println ( timestamp );  Serial.println();
} // end readSeesawSoilSensorReadings
```

ladyada, 2021-04-13:
thanks please stick to 1.3 for now, we just did a big refactor so not terribly surprising that some things are buggy!

ladyada, 2021-04-28:
ok try 1.4.1 please!

leandrodisiuta, 2021-04-28:
FYI I was facing the same problem. My readings for capacitance were 23~25. I was almost returning the sensor to the shop where I bought it.
After installing the updated library (1.4.1) it seems to work well, with readings 300~1100. Thanks for the update.
I am using a Arduino Mega 2560 with Arduino IDE 1.8.10.

closed 2021-04-28 by ladyada

thorathome, 2021-05-05:
Just saw a new update 1.4.2. Thanks for keeping this library current, @ladyada .

leandrodisiuta, 2021-05-07:
What is the capacitance unit? Farad? 
I mean, when you say 200-1200 range, what these numbers represent?

MorganEzra, 2022-07-21:
Hi I have an issue with this sensor, i try the exemple from the library it work yesterday but now i have only "Error seesaw not found" do you have an idea what is the problem ?

Cordially