adafruit/Adafruit_Seesaw #39: Seesaw soil sensor not measuring capacitance; only temperature

juliant99, 2021-04-20:
I used the seesaw soil sensor example which comes with version 1.4.0 of the soli sensor library and the capacitance logged is the same as the temperature. Ie if the logged temperature is, for example, 25.87*C the capacitance is 25 etc.

Reverting to version 1.3.1 of the seesaw library the capacitance is within the range expected, ie with the sensor in the air it is greater than 300.

The sensor is connected as per https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/arduino-test

ladyada, 2021-04-27:
what board are you using to read the sensor? we cannot reproduce this issue :/

juliant99, 2021-04-27:
I'm using the Arduino Uno - R3 .

ladyada, 2021-04-28:
ok try 1.4.1 please!

closed 2021-04-28 by ladyada

juliant99, 2021-05-08:
Yep, works now - many thanks