adafruit/Adafruit_Seesaw #47: Adafruit_seesaw.h does not work with Arduino IOT Cloud Generated Code

colinpwheeler, 2021-07-05:
There is a problem as soon as one uses Adafruit_seesaw.h with Arduino code generated by the Arduino IOT Cloud which includes:
#include <ArduinoIoTCloud.h>
#include <Arduino_ConnectionHandler.h>

Here is my code:
Main Code:
#include "arduino_secrets.h"
#include "thingProperties.h"
#include "Adafruit_seesaw.h"
Adafruit_seesaw ss;
void setup() {
  Serial.begin(115200);
  Serial.println("seesaw Soil Sensor example!");
  if (!ss.begin(0x36)) {
    Serial.println("ERROR! seesaw not found");
    while(1);
  } else {
    Serial.print("seesaw started! version: ");
    Serial.println(ss.getVersion(), HEX);
  }
  initProperties();
  ArduinoCloud.begin(ArduinoIoTPreferredConnection);
  setDebugMessageLevel(4);
  ArduinoCloud.printDebugInfo();
}
void loop() {
  float tempC = ss.getTemp();
  uint16_t capread = ss.touchRead(0);
  int moist = map(capread, 338, 570, 0, 100);
  Serial.print("seesaw started! version: ");
  Serial.println(ss.getVersion(), HEX);
  Serial.print("Temperature: "); Serial.print(tempC); Serial.println("*C");
  Serial.print("Capacitive: "); Serial.println(capread);
  Serial.print("Moist: "); Serial.println(moist);
  Serial.println();
  delay(1000);
}

thingProperties.h:
// Code generated by Arduino IoT Cloud, DO NOT EDIT.
#include <ArduinoIoTCloud.h>
#include <Arduino_ConnectionHandler.h>
const char THING_ID[] = "d0d57085-4094-465b-9743-2d2a320e1396";
const char SSID[]     = SECRET_SSID;    // Network SSID (name)
const char PASS[]     = SECRET_PASS;    // Network password (use for WPA, or use as key for WEP)
int pallas_Ping;
float pallas_i2c_Temp;
int pallas_ic2_Moisture;
void initProperties(){
  ArduinoCloud.setThingId(THING_ID);
  ArduinoCloud.addProperty(pallas_Ping, READ, ON_CHANGE, NULL);
  ArduinoCloud.addProperty(pallas_i2c_Temp, READ, ON_CHANGE, NULL);
  ArduinoCloud.addProperty(pallas_ic2_Moisture, READ, ON_CHANGE, NULL);
}

WiFiConnectionHandler ArduinoIoTPreferredConnection(SSID, PASS);

The correct output is when the line "ArduinoCloud.begin(ArduinoIoTPreferredConnection);" is commented out and is:
23:30:26.618 -> seesaw started! version: FBA278F
23:30:26.618 -> Temperature: 26.17*C
23:30:26.618 -> Capacitive: 346
23:30:26.618 -> Moist: 3

As soon as this line is enabled the output becomes:
23:36:37.945 -> seesaw started! version: 19000000
23:36:37.945 -> Temperature: 25.64*C
23:36:37.945 -> Capacitive: 65535
23:36:37.945 -> Moist: 28102

ladyada, 2021-07-05:
maybe the i2c port is unavailable? seesaw really just uses i2c and thats it, we dont have anything with arduinocloud so not sure what it is. you could ask arduinoiot if they have any suggestion :)

closed 2021-07-05 by ladyada

colinpwheeler, 2021-07-05:
The port seems to be available as it is giving me a temperature reading from that port.  It is your product that I am using and your suggested library that I am using.

ladyada, 2021-07-05:
that value is what happens when the i2c data is not coming through right, the version isnt right either. its something messed up with the i2c connectivity

colinpwheeler, 2021-07-05:
> that value is what happens when the i2c data is not coming through right, the version isn't right either. its something messed up with the i2c connectivity

Yes, that is exactly what I understand.  It works perfectly on the same hardware but when I include the Arduino Cloud code, it goes wrong.  I have tried it on different Arduino boards as well and always the same result.  As I cannot debug the library as I am not a developer, it is kind of hard to figure out if your library is breaking when I use it with the IOT Cloud libraries or if they are breaking.

It makes it all a bit of a mystery.  The examples available on the Internet all point there being problems in one of the three libraries in question.

The question here is why are the IOT Cloud libraries causing your Seesaw library to break?