adafruit/Adafruit_Seesaw #60: nRF52840 (Sense) hangs on this line on the last retry when Trellis isn't connected

adrienp, 2021-12-15:
I'm not sure how this is happening, and I'm curious if anyone else is able to reproduce this. I'm using the basic example sketch, compiled and uploaded with platformio. I have a kit 4x4 NeoTrellis setup and the trellis connector got disconnected, and execution got hung in `trellis.begin()`, so I added a bunch of debug prints and narrowed it down to the `delay(10);` on this line: https://github.com/adafruit/Adafruit_Seesaw/blob/b2d94a60a5537646913357d4615531ff7b477e82/Adafruit_seesaw.cpp#L80

It goes through all 10 retries and returns `false` just fine after the first call to `pixels.begin(...)` here: https://github.com/adafruit/Adafruit_Seesaw/blob/master/Adafruit_NeoTrellis.cpp#L30

But then it hangs on this `delay(10)` on the 10th retry when called from `Adafruit_seesaw::begin(...)` here: https://github.com/adafruit/Adafruit_Seesaw/blob/master/Adafruit_NeoTrellis.cpp#L32

I put a `Serial.printf("before delay, retry %d\n", retries);` before that delay and a `Serial.println("After retry loop");` after the loop, and the 9th retry is the last thing that's printed.

Curiously, if I put a `Serial.println("After delay");` right after the delay call inside the loop, THEN execution continues and the function returns! Which seems crazy.

Maybe this is some quirk with the `vTaskDelay` call in `delay()`?

ladyada, 2021-12-15:
if the i2c is disconnected, it will cause undefined behavior so while this is worth looking at...plz do not disconnect cables :)

adrienp, 2021-12-15:
Haha good to know! I just assumed that because of the following code in the basic example I should expect `trellis.begin()` to return and not hang even if the cable is disconnected, and then print the appropriate message.

```cpp
  if (!trellis.begin()) {
    Serial.println("Could not start trellis, check wiring?");
    while(1);
  } else {
    Serial.println("NeoPixel Trellis started");
  }
```

But interesting to know that if i2c is disconnected it could cause undefined behavior, thanks for explaining that!

ladyada, 2021-12-15:
@hathach will try to replicate when there's some time :) a lot i2c changed recently

hathach, 2021-12-15:
I will try to reproduce it and post update

hathach, 2021-12-16:
@adrienp could you confirm the issue still exists with Arduino IDE and also post your Arduino compile log as attached txt here for analysis. Since I only test and familiar with Arduino, I would like to make sure we are testing with the same setup as well as same version of BSP and used libraries.

adrienp, 2021-12-16:
@hathach Yes this issue also exists with Arduino IDE. When the trellis is properly connected, the "basic" example compiles and uploads and I get the expected message `NeoPixel Trellis started` in the Serial Monitor. When the trellis is not connected, it compiles and uploads, but nothing is printed to the serial monitor because it seems to hang on `trellis.begin()`.

Apologies for the long copy-pastes below, but here's my compile log:

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/adrien/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/adrien/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/adrien/Documents/Arduino/libraries -fqbn=adafruit:nrf52:feather52840sense:softdevice=s140v6,debug=l0,debug_output=serial -vid-pid=239A_8087 -ide-version=10816 -build-path /var/folders/cj/8v2cz6f54wz4xykm42mtk1bh0000gn/T/arduino_build_932577 -warnings=none -build-cache /var/folders/cj/8v2cz6f54wz4xykm42mtk1bh0000gn/T/arduino_cache_53179 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.CMSIS.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0 -prefs=runtime.tools.nrfjprog.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/nrfjprog/9.4.0 -prefs=runtime.tools.nrfjprog-9.4.0.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/nrfjprog/9.4.0 -verbose /Users/adrien/Documents/Arduino/libraries/Adafruit_seesaw_Library/examples/NeoTrellis/basic/basic.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/adrien/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/adrien/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/adrien/Documents/Arduino/libraries -fqbn=adafruit:nrf52:feather52840sense:softdevice=s140v6,debug=l0,debug_output=serial -vid-pid=239A_8087 -ide-version=10816 -build-path /var/folders/cj/8v2cz6f54wz4xykm42mtk1bh0000gn/T/arduino_build_932577 -warnings=none -build-cache /var/folders/cj/8v2cz6f54wz4xykm42mtk1bh0000gn/T/arduino_cache_53179 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.CMSIS.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0 -prefs=runtime.tools.nrfjprog.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/nrfjprog/9.4.0 -prefs=runtime.tools.nrfjprog-9.4.0.path=/Users/adrien/Library/Arduino15/packages/adafruit/tools/nrfjprog/9.4.0 -verbose /Users/adrien/Documents/Arduino/libraries/Adafruit_seesaw_Library/examples/NeoTrellis/basic/basic.ino
Using board 'feather52840sense' from platform in folder: /Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0
Using core 'nRF5' from platform in folder: /Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0
Detecting libraries used...
/Users/adrien/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=64000000 -DARDUINO=10816 -DARDUINO_NRF52840_FEATHER_SENSE -DARDUINO_ARCH_NRF52 "-DARDUINO_BSP_VERSION=\"1.2.0\"" -DNRF52840_XXAA -DUSBCON -DUSE_TINYUSB -DUSB_VID=0x239A -DUSB_PID=0x8087 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather nRF52840 Sense\"" -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DDX_CC_TEE -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -DCFG_LOGGER=0 -DCFG_SYSVIEW=0 -I/Users/adrien/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0/CMSIS/Core/Include/ -I/Users/adrien/Library/Arduino15/packages/adafruit/tools/CMSIS/5.7.0/CMSIS/DSP/Include/ -I/Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0/cores/nRF5/nordic -I/Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0/cores/nRF5/nordic/nrfx -I/Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0/cores/nRF5/nordic/nrfx/hal -I/Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0/cores/nRF5/nordic/nrfx/mdk -I/Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0/cores/nRF5/nordic/nrfx/soc -I/Users/adrien/Library/Arduino15/packages/adafruit/hardware/nrf52/1.2.0/cores/nRF5/nordic/nrfx/dr [... cut ...]
Upgrading target on /dev/cu.usbmodem146301 with DFU package /private/var/folders/cj/8v2cz6f54wz4xykm42mtk1bh0000gn/T/arduino_build_932577/basic.ino.zip. Flow control is disabled, Single bank, Touch disabled
Opened serial port /dev/cu.usbmodem146301
Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 47296
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
########################################
#############
Activating new firmware

Device programmed.
DFU upgrade took 3.2613608837127686s
```

hathach, 2021-12-17:
thank for the confirmation and log, please post it as attached txt, should you have to do it again

hathach, 2022-01-04:
@adrienp sorry for late response, I got my time to try this out. Device is not actually hanged, the `Serial.print()` does not appear because of `while(1) ;` blocking 100% CPU. TinyUSB stack run in the background and requires a delay()/yield() in order to send/receive data from/to USB bus. This is quite typical pitfall when using with TinyUSB. I will update the example to also add delay(1).  

Note: You could also force serial immediately by using `Serial.flush()`. 

```C
  if (!trellis.begin()) {
    Serial.println("Could not start trellis, check wiring?");
    while(1) delay(1);
  } else {
    Serial.println("NeoPixel Trellis started");
  }
```

adrienp, 2022-01-04:
Oh wow! Thanks for digging into this and responding @hathach !

hathach, 2022-01-04:
no problem at all, I am glad it makes sense to you.

PR #62 (2022-01-04): add delay(1) to while(1) error loop
add `delay(1)` to error loop while(1), this help to prevent user confusion
- fix #60
closed 2022-01-04 by hathach