adafruit/Adafruit_Seesaw #83: RP2040 Feather cannot interface to Mini TFT Wing unless Seesaw library code is manually changed.

emeb, 2023-05-23:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  RP2040 Feather  + miniTFTwing

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.18.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):

1) Connect RP2040 Feather to Mini TFT Wing
2) Install latest Seesaw library and dependencies
3) Install earlephilhower Arduino Pico board set (latest version 3.2.0)
4) Attempt to upload any Mini TFT Wing examples
5) Observe no proper operation
6) Modify seesaw base class to use Wire1 instead of Wire as default I2C port
7) Recompile
8) Observe proper operation.

The root cause appears to be that the RP2040 documentation implies that the Feather header SDA/SCL pins are attached to I2C port 0 (Wire/Wire0) and the Qwiic connector is on I2C port 1 (Wire1), where in reality they're both connected to port 1.

This issue has also been reported to the Arduino Pico project: https://github.com/earlephilhower/arduino-pico/issues/1465

caternuson, 2023-05-23:
This is a legacy "feature". When the Feather RP2040 was originally added to the BSP, it lacked the ability to "map" the I2C pins to `Wire` or `Wire1` as desired. So the pins being used by the Feather RP2040 ended up as `Wire1` since they use the `I2C1` peripheral of the RP2040.

Note that a non-default I2C bus can be passed in when creating the miniTFT seesaw instance. So the sketch's would (should) work with minor modification to the sketch code. No need to change any library code. Change:
```cpp
Adafruit_miniTFTWing ss;
```
to:
```cpp
Adafruit_miniTFTWing ss(&Wire1);
```

I2C being `Wire1` is mentioned here:
https://learn.adafruit.com/adafruit-feather-rp2040-pico/pinouts#stemma-qt-3084853
but confusing since it's about the STEMMA QT connector. But the header pins are the same.

emeb, 2023-05-23:
OK - looks like this is a non-conforming interface that can't be fixed without breaking all existing code. Perhaps the RP2040 feather documentation could be extended to mention your suggested instantiation change above?
At any rate, I'll close this issue now.

caternuson, 2023-05-23:
Going to do both. Going ahead with making the breaking change:
https://github.com/earlephilhower/arduino-pico/pull/1468
and will update documentation - mainly to cover that it will now be `Wire` (or default) instead of `Wire1`.

This will make things easier in the long run. No need to make the explicit reference to `Wire1`, which has been an ongoing source of confusion and issues.