adafruit/Adafruit_Seesaw #84: Attiny1616 "seesaw not found" with Arduino Nano 33 BLE

silvan2468, 2023-05-24:
Hy
I really like your products and have a lot of them. 

But unfortunately, there is a problem at the attiny1616 (together with Arduino Nano 33 BLE):
- when trying the blink example --> error, seesaw not found
- I2C scanner: no device found

- when I do exactly the same with a Seeeduino Xiao, both works (scanner and blink demo)
- when I scan on the Nano 33 BLE on Wire1, 2x I2C sensors on the Arduino Nano 33 BLE are shown with address
- when I scan with the Nano for the sparkfun BME680, it works, too.
- when I scan with the Nano for your Stemma GPS-module, addresse 0x10 is shown, too.

Just the seesaw Attiny1616 is not detected.
I hope you find a solution before the weekend. 

Thanks and kind regards
Silvan

caternuson, 2023-05-24:
Are you using one of these breakouts?
https://www.adafruit.com/product/5690

If so, how is it being connected to the Arduino board?

Is the green LED on the breakout coming on?

What I2C scanner sketch is being used?

Is the Arduino Nano 33 BLE your only Arduino board?

silvan2468, 2023-05-25:
Wow. Fast answer. Thanks :)

Yes, I use your 5690.

I connect as follows on a bread board:
- 3.3V from Arduino to Vin on 5690
- GND to GND
- SDA and SCL with wires to SDA and SCL

When I connect your Stemma Mini GPS (4415) over the Stemma connector to the 5690, the I2C scanner from your site finds the GPS module. So connection of I2C looks good.

I tested it right now also with an Arduino 33 IoT and there it works without problem.

I tested also all 3 other combinations of addresses by pulling pins 12 and 13 low because I thaught the address 0x49 is perhaps wrong with the Arduino Nano 33 BLE, but it works not with all 4 addresses.

Probably a problem with the MBED from Arduino 33 BLE? The Arduino 33 IoT and the Xiao don't use MBED.

I2C scanner: I copied this code from your website 
https://learn.adafruit.com/scanning-i2c-addresses/arduino
to the Arduino IDE without modification:

```
// --------------------------------------
// i2c_scanner
//
// Modified from https://playground.arduino.cc/Main/I2cScanner/
// --------------------------------------

#include <Wire.h>

// Set I2C bus to use: Wire, Wire1, etc.
#define WIRE Wire

void setup() {
  WIRE.begin();

  Serial.begin(9600);
  while (!Serial)
     delay(10);
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    WIRE.beginTransmission(address);
    error = WIRE.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknown error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```

silvan2468, 2023-05-25:
I found this:
https://forum.arduino.cc/t/no-i2c-devices-found-nano-33-ble-ms5611-mkr-gps-shd/942515/22
and then this:
https://github.com/RobTillaart/MS5611/issues/31

Looks my feeling with problem with MBED was right :(

When I include this line to the scanner-code above:
WIRE.write(0x00);

So it looks like:
...
    WIRE.beginTransmission(address);
    WIRE.write(0x00);
    error = WIRE.endTransmission();
...

Than the Attiny1616 is found!

As the MBED problem is already one year old and not solved yet (I am on the newest MBED core version), do you see a solution  on your side at the seesaw-code to solve the problem?

Further it is curious for me, why the GPS module is found and the Attiny1616 not. 

Thanks a lot.
Silvan

caternuson, 2023-05-25:
Good info and sleuthing. Looks like you've found the underlying issue. The proper solution would be for the MBED core to fix this, instead of adding preprocessor patches and work arounds elsewhere.

Some other discussion here:
https://forum.arduino.cc/t/nano-33-ble-i2c-problems-caused-in-mbed-wire-cpp-read-write-bit-on-i2c-scan/964676/7
which has a reference to this open issue:
https://github.com/arduino/ArduinoCore-mbed/issues/414
That'd be the issue thread to monitor for progress with fixing this in the core.

silvan2468, 2023-05-25:
Hy caternuson
Thx for the answer. I understand you, that you are not willing to solve other problems with your code.

BUT I thaught a bit today about this and if it is really a problem from mbed and not from seesaw: why does the Arduino Nano 33 BLE detect 3 different I2C-sensors and even your own gps module, but only not the seesaw?!

By the way: should the I2C to UART-bridge been working with the ATtiny1616 (5690)?
I think this is not (yet) clear written on your homepage. I tried it for some minutes, but could not get it working to send an echo (Tx and rx of 5690 shorted) and read it out over I2C with the arduino.

caternuson, 2023-05-25:
The I2C interaction done in the scan, and also in the `detected()` method of busio, is a sort of hackish. The I2C spec doesn't really provide for this sort of usage. So results can vary. For example, the MBED core is also causing a similar issue with the MS5611 pressure sensor referenced in the other issue thread linked above.

Can you link to where I2C to UART is mentioned?

silvan2468, 2023-05-25:
I like to use the attiny1616 also to have one more uart channel. I tried more or less this example:
https://github.com/adafruit/Adafruit_Seesaw/blob/master/examples/communication/UART_loopback/UART_loopback.ino

But only with:

```
....
ss.print(0x02)
Delay(100);
char c = ss.readSercomData();
Serial.print(c);
....
"

caternuson, 2023-05-25:
You'll need to build and update the firmware on the ATtiny1616 to include this PR:
https://github.com/adafruit/Adafruit_seesawPeripheral/pull/10

The firmware shipped on the breakout you have does not include that. (keep in mind the PID 5690 is mainly a developer board)

silvan2468, 2023-05-25:
Ok. I will order your usb-to-serial programmer to reprogram the attiny1616.

Thanks a lot. Topic is not proper solved as long mbed is not modyfied, but now clear to me what the problem is.
I close now.

closed 2023-05-25 by silvan2468