adafruit/Adafruit_Seesaw #90: setEncoderPosition() not working

tomcostanza, 2023-09-11:
I'm using the demo code from the learning site. If I put:

for (int i = 0; i < 4; i++) {
	  ss.setEncoderPosition(50, i);
  }
  for (int i = 0; i < 4; i++) {
	  Serial.println(ss.getEncoderPosition(i));
  }

at the end of the setup code, I print 4 zeros. I don't think it's a bad board, because everything else seems to work.

cnc4less, 2023-09-17:
The same problem still waiting for an answer

caternuson, 2023-09-17:
What seesaw based product is being used?

cnc4less, 2023-09-17:
Adafruit I2C Quad Rotary Encoder Breakout with NeoPixel - STEMMA QT / Qwiic

tomcostanza, 2023-09-17:
> The same problem still waiting for an answer

Adafruit product number 5752, running on an Arduino Uno R3

cnc4less, 2023-09-18:
Adafruit product number 5752, running on an Arduino  esp32-s3 still the same problem.

caternuson, 2023-09-18:
Sorry about this, but it looks like the issue is with the current firmware run on the ATtiny based seesaw's. It does not have support for setting encoder position. Only reading. Opened a new issue for that here:
https://github.com/adafruit/Adafruit_seesawPeripheral/issues/20

Options for your current board:
* Handle the same functionality in user code
* Reprogram them once updated firmware becomes available ([see here for how](https://learn.adafruit.com/adafruit-attiny817-seesaw/advanced-reprogramming-with-updi))
* Can refund if none of these options will works (for that, please [post in forums](https://forums.adafruit.com))

tomcostanza, 2023-09-19:
> Sorry about this, but it looks like the issue is with the current firmware run on the ATtiny based seesaw's. It does not have support for setting encoder position. Only reading. Opened a new issue for that here: [adafruit/Adafruit_seesawPeripheral#20](https://github.com/adafruit/Adafruit_seesawPeripheral/issues/20)
> 
> Options for your current board:
> 
> * Handle the same functionality in user code
> * Reprogram them once updated firmware becomes available ([see here for how](https://learn.adafruit.com/adafruit-attiny817-seesaw/advanced-reprogramming-with-updi))
> * Can refund if none of these options will works (for that, please [post in forums](https://forums.adafruit.com))

1. Can I assume the same problem exists with Adafruit 4991.  I have several of those but have not tried them yet.
2. Is there an option of returning the boards to you and have you reprogram them instead of a refund?

Thanks,
-Tom

caternuson, 2023-09-19:
PID 4991 uses different seesaw firmware (for SAMD), so should be OK. Or, if there is an issue, would be something different.

However, PID 5221 would have this same issue:
https://www.adafruit.com/product/5221

Unfortunately, there is no option to return the items for reprogramming.

wim-number37, 2023-09-23:
I can confirm that PID 5740 (Adafruit ANO Rotary Navigation Encoder to I2C QT Adapter) has this issue.

ScottFerg56, 2023-10-07:
I just encountered this today. I can confirm that the single rotary encoder, 4991, works fine. I decided to expand to the Quad encoder, 5752, and discovered that setEncoderPosition() does not work. A day's work lost, but pressing on with a software workaround, if possible!

caternuson, 2024-02-27:
Closing this issue since it is resolved in terms of updating repo code. The actual issue lies in a different repo:
https://github.com/adafruit/Adafruit_seesawPeripheral/issues/20
and has been fixed via this pull request:
https://github.com/adafruit/Adafruit_seesawPeripheral/pull/21

**There is no actual issue with this repo.**

For anyone receiving hardware programmed with older seesaw firmware that does not support the seesaw set encoder position, options are generally as stated above:
* Handle the same functionality in user code (see example below)
* Reprogram them once updated firmware becomes available ([see here for how](https://learn.adafruit.com/adafruit-attiny817-seesaw/advanced-reprogramming-with-updi))
* Can refund if none of these options will works (for that, please [post in forums](https://forums.adafruit.com))

Here's some example stub code for what adding functionality in user code would look like: 
```cpp
int32_t encoder_offset = 0;

void setEncoderPosition() {
  encoder_offset = ss.getEncoderPosition();
}

int32_t getEncoderPosition() {
  return ss.getEncoderPosition() - encoder_offset;
}
```

closed 2024-02-27 by caternuson