adafruit/Adafruit_Seesaw #93: Cannot change I2C address of the Adafruit Mini I2C STEMMA QT Gamepad

pongsagon, 2023-10-25:
In the document, one should be able to change the I2C address by calling setI2CAddr().
The default address of Adafruit Mini I2C STEMMA QT Gamepad is 0x50.

However, I have try using  setI2CAddr() as well as try calling  EEPROMWrite8(SEESAW_EEPROM_I2C_ADDR, newAddress) directly and then call ss.begin(newAddress).

Calling ss.begin(newAddress) would return false. 
The function can actually store the new address to EEPROM.  I have verified this by calling getI2CAddr(), even after the replug the power cable.
But, the I2C address that can used to ss.begin() must still be the old one, 0x50.

Look like the gamepad does not use or care the I2C address from EEPROM.

more detail on this post https://forums.adafruit.com/viewtopic.php?t=205373 

Thanks a lot

Matt

caternuson, 2023-10-25:
Was able to recreate this. The issue is the hardwired value of `0x3F` for `SEESAW_EEPROM_I2C_ADDR` here:
https://github.com/adafruit/Adafruit_Seesaw/blob/c3e7b8f4dfdcc1f8ca3c0cabbacfd441ba8f8212/Adafruit_seesaw.h#L202-L204

That's the correct location for SAMD based seesaw firmware:
https://github.com/adafruit/seesaw/blob/af364d5bcd24330fbb00d9c34e6b27f090af7ab3/include/RegisterMap.h#L215

However, for ATtiny based seesaw firmware, the location is the end of EEPROM, which is determined dynamically:
https://github.com/adafruit/Adafruit_seesawPeripheral/blob/12dd2e85d294eee182922d6cc544f17bb30ef19a/Adafruit_seesawPeripheral.h#L26
The EEPROM address location ends up being `0x7F` on ATtiny.

So this library is reading/writing to the incorrect  EEPROM location for any ATtiny based seesaw product.

pongsagon, 2023-10-26:
Thanks I can change the I2C address now.

 Adafruit_seesaw ss;  
 ...                                                                                                                                                 
uint8_t newAddress = 0x49;                                                                                                                    
ss.EEPROMWrite8(0x7F, newAddress);   
delay(250);
ss.begin(newAddress);      // now return true and can read the input

closed 2023-10-26 by pongsagon