adafruit/Adafruit_Seesaw #97: see_saw library doesn't compile for Arduino Nano ESP32

steffenmarkkus, 2024-03-16:
- Arduino board:  Nana ESP32 S3

- Arduino IDE version (found in Arduino -> About Arduino menu):  Version: 2.3.2
Date: 2024-02-20T09:57:46.613Z
CLI Version: 0.35.3

Copyright © 2024 Arduino SA

- List the steps to reproduce the problem below

1) Installed Adafruit SeeSaw Library ver 1.7.5
2) Connect 1x4 encoder to SDA/SCL, pins a4, a5, and to power and ground
3) Load and compile LEDArcade_1x4.ino

Compiler responds with the following:  

In file included from /private/var/folders/n3/wpnkdym15vv8970_qdlwknvw0000gn/T/.arduinoIDE-unsaved2024215-50002-16ik41b.sypz/LEDArcade_1x4/LEDArcade_1x4.ino:5:
/Users/srosemer/Documents/Arduino/libraries/Adafruit_seesaw_Library/Adafruit_seesaw.h:253:74: error: macro "analogWrite" passed 3 arguments, but takes just 2
   virtual void analogWrite(uint8_t pin, uint16_t value, uint8_t width = 8);
                                                                          ^
exit status 1

caternuson, 2024-03-16:
This is a known issue with the ESP32 Arduino Board Support Package:
https://github.com/espressif/arduino-esp32/issues/9150

There's a suggested work around here (see "easy out" way):
https://github.com/espressif/arduino-esp32/issues/9150#issuecomment-1903804980

steffenmarkkus, 2024-03-16:
> espressif/arduino-esp32#9150
> espressif/arduino-esp32#9150 (comment)

Thanks Carter, that seems to do the trick.

Steve
SteffenMarkkus Specialty Roasters
(424) 308-5187

closed 2024-05-03 by caternuson