adafruit/Adafruit_Seesaw #98: servo

tracton, 2024-03-20:
- Hardware:  ESP32 to ATtiny1616 over I2C; ATtiny is supposed to output PWN (with 5-10% duty cycle) on pin 11 to Motor driver that drives a Servo SM-S2309S.  Oscilloscope on pin 11.
- Arduino version: 2.3.2

**BUG:** Adafruit_seesaw.cpp appears to have a bug in that the chip is not identified in the `Adafruit_seesaw::setPWMFreq` function, and so legit pins can not have their PWM frequency set; the call is prematurely returned.  

**SOLUTION:** The `setPWMFreq `function should look like the `analogWrite` function which tests each Hardware ID Code to determine which pins support PWM per chip.  I think it should look a bit like this:

```void Adafruit_seesaw::setPWMFreq(uint8_t pin, uint16_t freq) {
  int8_t p = -1;
  if (_hardwaretype == SEESAW_HW_ID_CODE_SAMD09) {
    switch (pin) {
    case PWM_0_PIN:
      p = 0;
      break;
    case PWM_1_PIN:
      p = 1;
      break;
    case PWM_2_PIN:
      p = 2;
      break;
    case PWM_3_PIN:
      p = 3;
      break;
    default:
#ifdef SEESAW_I2C_DEBUG
      Serial.printf("SEESAW: PWM not supported on Pin %d for hardware %d\n", pin, _hardwaretype);
#endif
      break;
    }
  } else {
    p = pin;
  }

caternuson, 2024-03-20:
Hmm. Agree. Looks like that logic needs updating for the ATtiny variants, like you mention - similar to what's in `analogWrite`.

Thanks for changing the issue title.

caternuson, 2024-05-03:
Closing. Should be fixed with #99.

closed 2024-05-03 by caternuson