adafruit/Adafruit_Seesaw #101: ESP32 and I2C encoder

yo2ldk, 2024-06-13:
I buy this 4 encoders board, to use with an ESP32 who have all pins busy with parallel TFT display
 when I use your example 

 "#define SS_SWITCH        24
   #define SS_NEOPIX        6",
   
   this block my display because ESP32 take HIS 24 and 6 pin, not from I2C board
   how can I define to be ok ??
   please made an example how to use your library with ESP..

caternuson, 2024-06-13:
> I buy this 4 encoders board

What specific product is this referring to?

yo2ldk, 2024-06-13:
> What specific product is this referring to?
[Adafruit I2C Quad Rotary Encoder Breakout] and ESP32 M5 Atom Lite

caternuson, 2024-06-13:
Is this a separate issue from #102?

yo2ldk, 2024-06-13:
same thing sorry,
firs time I see it not work as I supposed and should,
I not find anywhere how to declare pin A and B, etc
then I check hardware, I see is not found on I2C..
but anyway, is two apart things

caternuson, 2024-06-13:
Is it the same? Or is it two separate issues? Both are being stated. So it's confusing. It's unclear what *this* issue is about? It seems like this is a basic connection issue and it is the same as being dealt with in #102.

yo2ldk, 2024-06-14:
ok, both are opened by me
in this issue, I want  to clarify how can be declared for an ESP2, the Pin A or B on each of the four encoders, not just push button as in all your examples. I don't need rainbows, joysticks or other LED shows, I need to use as name is, for encoders via I2C, because I not have enough IO ports on the M5 Atom Lite ESP module, and I cannot use other for my project..
the page for Adafruit I2C Quad Rotary Encoder  Pinouts are very clear with good description, but examples for encoders are...
the#102 issue is for not recognised board on I2C.

caternuson, 2024-06-14:
OK, thanks for clarifying. I get the question now.

Those pins are fixed in the firmware, which is built specific for the Quad Rotary Encoder product (PID 5752):
https://github.com/adafruit/Adafruit_seesawPeripheral/blob/8b2fcaa1859b19a3ed037bdf96d9a8a942528308/examples/example_pid5752/example_pid5752.ino#L19-L26
They can not be changed or reassigned via I2C.

Are you wanting a general purpose I2C-to-encoder solution? You do not actually want to use the PID 5752?

yo2ldk, 2024-06-16:
Thank you for the link and help !
This is not in page product description.. I will try
my best regards!

closed 2024-06-16 by yo2ldk