adafruit/Adafruit_Seesaw #102: seewaw not find it on default address..

yo2ldk, 2024-06-13:
looking further, I find that my ESP32 board, using YOUR example - encoder_basic- , not find your I2C encoder 
<
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Looking for seesaw!
Couldn't find seesaw on default address
>
even if I declare it
#include "Adafruit_seesaw.h"
#include <seesaw_neopixel.h>

#define SS_SWITCH        12
#define SS_NEOPIX        6
#define SEESAW_ADDR          0x49
where is declared PIN A and PIN B in ALL your examples ??
once again,on ALL your example, I find just for switch,neopix led,and all possible, EXCEPT  a good example for ENCODER, because i buy it just for that: to use it as I2C ENCODER !!  as is name you give it..

caternuson, 2024-06-13:
What happens if you run the example from the guide?
https://learn.adafruit.com/adafruit-i2c-quad-rotary-encoder-breakout/arduino

yo2ldk, 2024-06-13:
> What happens if you run the example from the guide? https://learn.adafruit.com/adafruit-i2c-quad-rotary-encoder-breakout/arduino

  Thank you for fast reply,
  this is problem, nothing happens, declared as:       #define SEESAW_ADDR      0x49,
  on serial have follow message:
  
  "configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Looking for seesaw!
Couldn't find seesaw on default address
"

caternuson, 2024-06-13:
Does the address show up in an I2C scan?
https://learn.adafruit.com/scanning-i2c-addresses/arduino

yo2ldk, 2024-06-13:
it is present at 0x49
[image]

caternuson, 2024-06-13:
Weird. It's showing up in the I2C scan as expected. So unclear why the example code is not working.

Is the M5Stack Atom Lite your only Arduino board?

yo2ldk, 2024-06-14:
no, I have also others boards, I will try when I have time but, 
I need for Atom Lite because not have enough IO ports,
 more exactly is Atom lite HDMI display, and all his ports is for that driver

yo2ldk, 2024-06-14:
ok, this issue will be closed, I found the problem
If your I2C encoders boards is not recognised on I2C, 
just add on void setup the line
"Wire.begin(25, 21);"  where 25,21 is YOUR  xx, xx arduino board SDA, SCL pinout numbers.

closed 2024-06-14 by yo2ldk